EricHigdon/OctoPrint-RGB_status #1: LEDs

codegrue, 2019-02-28:
Thanks for making this. Do you have any references for where to get the LED's to control?

EricHigdon, 2019-02-28:
Most of my LEDs have come from Amazon. The Alitove LEDS always seem to be fairly good quality, so maybe [these](https://www.amazon.com/ALITOVE-Individual-Addressable-Programmable-Non-Waterproof/dp/B01MG49QKD/ref=sr_1_5?keywords=ws2811+led+strip+5v&qid=1551381721&s=gateway&sr=8-5).

However, this plugin should support anything that rpi_ws281x supports, so the options are virtually endless. I've also done some testing with LED pixels that I got from [Ray Wu's Aliexpress store](https://www.aliexpress.com/store/701799).

closed 2019-02-28 by EricHigdon

codegrue, 2019-02-28:
Perfect, Thank you!

codegrue, 2019-02-28:
The Amazon one will plug right into the Raspberry Pi circuit board? Forgive me I'm new to this stuff.

EricHigdon, 2019-02-28:
For the most part. You'll probably want to use an external power supply (your 3D printer supply may work if you use a buck converter to shift the voltage).

The [Adafruit tutorial](https://learn.adafruit.com/neopixels-on-raspberry-pi) for Neopixels on the Raspberry Pi is a very good resource for getting started. 

For the most part, [this image](https://learn.adafruit.com/assets/63928) shows the wiring that you will need. The main difference is that you'll probably want to use pin 10 since pin 18 (used in the image) requires that you run OctoPrint as root.