EricHigdon/OctoPrint-RGB_status #3: After Installing RGB-Status Octoprint crashes on print start

foxabilo, 2019-03-01:
After Installing RGB status from the plugins menu in octoprint with OctoPrint 1.3.10 running on OctoPi 0.16.0 Octoprint crashes when a print starts.

EricHigdon, 2019-03-01:
@foxabilo Can you share some more info? Maybe your [octoprint.log](https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299)?

foxabilo, 2019-03-01:
[logs.zip](https://github.com/EricHigdon/OctoPrint-RGB_status/files/2920926/logs.zip)

foxabilo, 2019-03-01:
Anything I can do to help just ask

EricHigdon, 2019-03-01:
@foxabilo Interesting. I'm seeing that rpi_ws281x is having trouble initializing.

Do you have SPI enabled on your raspberry pi? 
`sudo raspi-config` Interfacing Options > SPI

foxabilo, 2019-03-01:
It was enabled prior to this, but then I formatted the SSD to install OctoPi 0.16.0 so it will be whatever the default status for SPI is.

foxabilo, 2019-03-01:
the option screen does not tell me what it is, just if I would like to change it

EricHigdon, 2019-03-01:
@foxabilo Maybe make sure the pi user is in the gpio group?

`sudo adduser pi gpio`

foxabilo, 2019-03-04:
ok, i'll give it a try

tedder, 2019-03-06:
Coming here to see if you were aware of the octoprint community forum, Eric, seeing the same error. Can you catch the error and fail gracefully, at a minimum?

Here's the user's log and a link to the discussion topic where you can see the difficulty of finding it.
```
2019-03-05 21:44:35,919 - octoprint.plugin - ERROR - Error while calling plugin rgb_status
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 230, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_rgb_status/__init__.py", line 120, in on_after_startup
    self.init_strip()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_rgb_status/__init__.py", line 111, in init_strip
    self.strip.begin()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 124, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -13 (Unable to initialize SPI)
```

and the thread: https://community.octoprint.org/t/server-quits-when-i-hit-print/8092/6

EricHigdon, 2019-03-06:
@tedder I am aware of the forum, but had not seen that thread. Thanks for the heads up. I pushed an update that will ensure initialization problems like this don't cause OctoPrint to crash.

foxabilo, 2019-03-06:
Sorry people, I am not being much help at the moment, I have a back log of prints to get done for customers, as soon as they are finished I will re-install and retest with the sudo adduser pi gpio

EricHigdon, 2019-03-06:
@foxabilio no worries. I’m fairly confident that the group thing is your issue as I was able to replicate the problem on my instance. 63d3ab2 Should also have a fix to fail gracefully and log the error when this happens.

closed 2019-03-08 by EricHigdon