EricHigdon/OctoPrint-RGB_status #8: LEDs won't turn on.

marrobHD, 2019-03-10:
My WS2812B, WS2812 and WS2811 LEDs won't turn on.

EricHigdon, 2019-03-10:
@marrodHD can you share your octoprint.log? That should give me some insight into the cause of the issue.

marrobHD, 2019-03-10:
https://jpst.it/1DwHR

EricHigdon, 2019-03-10:
2019-03-10 17:57:14,555 - octoprint.plugins.rgb_status - ERROR - ws2811_init failed with code -11 (Selected GPIO not possible)

Seems relevant. What do you have the LED Pin setting set to?

marrobHD, 2019-03-10:
GPIO 13

marrobHD, 2019-03-10:
That are my settings:
[image]

EricHigdon, 2019-03-10:
@marrobHD Try setting LED channel to 1.

marrobHD, 2019-03-10:
I have that before it won't work. Now the LEDs won't work like before.

EricHigdon, 2019-03-11:
Are you running OctoPrint as root?

marrobHD, 2019-03-11:
I have installed octoprint on the sdcard directly. I have used the octoprint image.

marrobHD, 2019-03-11:
I think that it is root right? I can use pin 10.

EricHigdon, 2019-03-11:
OctoPrint does not run as root by default, and it is not recommended that you do so. If you have no other reason to use a different pin, I highly suggest sticking to 10.

marrobHD, 2019-03-11:
How can I change that

marrobHD, 2019-03-11:
It looks like the raspberry don't have a gpio13
[image]

EricHigdon, 2019-03-11:
@marrobHD You probably shouldn't try to run OctoPrint as root ([see here](https://community.octoprint.org/t/starting-octoprint-with-sudo-as-root-tells-me-you-should-not-run-octoprint-as-root-why/230)). 

In the image you attached, 13 would be the 4th pin from the bottom in the left column as we are looking at the BCM number.

I highly suggest switching to pin 10 since you wouldn't need to run as root. If pin 10 isn't working, make sure that you add the pi user to the gpio group, and enable SPI ([see here](https://github.com/EricHigdon/OctoPrint-RGB_status#running-without-root)).

You also may need to change your buffer size and core frequency ([see here](https://github.com/jgarff/rpi_ws281x#spi)).

If it still isn't working after following those steps, please share your octoprint.log again after rebooting.

I'm working on adding a wizard to walk new users through this stuff since it can be complicated.

marrobHD, 2019-03-11:
MOSI is pin 10 right?

EricHigdon, 2019-03-11:
@marrobHD Yes.

cryoplay, 2019-03-11:
Hello there.

I have the same Problem.
Hocked up my Strip to Pin 10.
Nothing Happens. Process turns on without a Problem but i had no light.

https://justpaste.it/6wnmc

EricHigdon, 2019-03-11:
@cryoplay From your log, it looks like everything is actually working on the software side. Do you have your LEDs attached to pin 10 and the ground pin on the pi?

marrobHD, 2019-03-12:
Yes

cryoplay, 2019-03-12:
Hi Eric.

My Problem is gone.
I forgot to hook the Strip to the ground of the Pi.
Sry. im really dump.

Now all is working perfectly.

EricHigdon, 2019-03-12:
@marrobHD can you share your OctoPrint.log again?

marrobHD, 2019-03-12:
Here is my log: https://jpst.it/1DHLe

EricHigdon, 2019-03-12:
@marrobHD It still says the "Selected GPIO not possible" are you sure you changed it to pin 10 in the settings and restarted?

marrobHD, 2019-03-12:
Yes

EricHigdon, 2019-03-13:
Is the spidev device visible on your machine?
Check using:
ls /dev/spidev0.*

marrobHD, 2019-03-13:
It is now working. I had to remove PSU control...

closed 2019-03-13 by EricHigdon