EricHigdon/OctoPrint-RGB_status #12: Only 1st LED shows Print progress

NeKoMuck, 2019-03-15:
While using the Print Progress Function, only the first LED of my strip is changing the colour. All other 59 LED still using the base color.

All other Functions of the Plugin seems to be working.

EricHigdon, 2019-03-15:
@nekomuck can you please share your OctoPrint.log file?

NeKoMuck, 2019-03-15:
@EricHigdon Here it is...

[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/2972508/octoprint.log)

NeKoMuck, 2019-03-16:
I'am using a WS2812B Stripe. LED type is set to WS2812, because there is no WS2812B option in your plugin.

Maybe this is the problem ?

The octoprint.log seems to count more than the 60 LEDs, i have set. Is that correct ?

I have made a complete new installation of Octoprint and RGB Status, but the problem is still available.

EricHigdon, 2019-03-16:
Ws2812 should be fine. If the other effects are working, that’s certainly not the problem. 

I’d guess it’s a problem with my algorithm. You have 60 leds? What happens if you set the number of leds to 10 in the settings?

EricHigdon, 2019-03-16:
I've confirmed that this is not working properly for larger numbers of LEDs.

NeKoMuck, 2019-03-16:
Thank you for your Feedback. Good to know, its not my fault.

Using only 10 LED works perfect. First time, that the second LED is changing the color.

I hope you will find the problem. I appreciate your plugin.

EricHigdon, 2019-03-16:
This should be fixed with 7cfe918. Once I’ve done some more testing, I’ll release.

NeKoMuck, 2019-03-16:
It would be great, when the LED turn off after a (self)defined time, when the Job has finished.

EricHigdon, 2019-03-16:
@NeKoMuck when the print completes, the leds go back to the idle effect, so you could accomplish something like that by setting the idle effect to a solid color and the color to black.

NeKoMuck, 2019-03-16:
I set the idle color to solid black, but the LED still shining green. I will try it again tomorrow.

[image]

EricHigdon, 2019-03-16:
Fixed in 7cfe918

closed 2019-03-16 by EricHigdon