EricHigdon/OctoPrint-RGB_status #14: Please support sending colors using M150 (Marlin)

coldtobi, 2019-03-17:
My LED strip is connected on my FYSETC, so it would be cool to control that one...
Of course, animations are hard there, but well, they are not that important.
Thanks!

EricHigdon, 2019-03-17:
@coldtobi This would be awesome. I wish I had a way to test. I may be able to do this if You or someone else can volunteer to test for me.

coldtobi, 2019-03-18:
I can help testing that. Just ping me ;-)

AGithub457, 2019-05-18:
@EricHigdon I can test this, I am currently in the process of turning a massive print farm into LED stripped ones, this will help a lot!

JohnTRoth, 2020-07-07:
I'd also be happy to test, and would also add that it would sure be nice to trigger desired effects via G-code as well. Perhaps a M150 E<effect_name> option. 

My use case is an OctoScreen LCD display, which allows addition of buttons to trigger G-code on demand. It would sure be nice to be able to add buttons to enable different effects via the touchscreen.

JohnTRoth, 2020-07-08:
I took a first stab at this, and have it supporting R, G (or U), B, P, and I parameters, but one bug I haven't been able to track down... When using I to set an individual pixel, it clears all others and I have no idea why. Upon realizing the issue, I even added code to get the pixel color first and explicitly set it to that same color - but it still turns all others off. Hard-coding to a specific color works though, so I suspect the getPixelColor call is failing. Seems like a start though.

Also - I don't know how to contribute to your GitHub project. Still learning... In the meantime, the modified files are attached.

[OctoPrint RGB.zip](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4893658/OctoPrint.RGB.zip)

EricHigdon, 2020-07-09:
@JohnTRoth i think the issue you are seeing with the LEDs being reset ever time the M150 gcode is run is because ever time you start a new effect it kills the old sub process and reinitializes the whole strip in a new Sub process. If you want to have some sort of update functionality without killing the last effect, you should try something similar to the progress update that I am doing using the queue. When processing M150, just put the new data in the queue to be handled by the sub process.

JohnTRoth, 2020-07-09:
@EricHigdon I'm certain you're correct now that I look at the way you handle it... I'm not familiar with the queue functions though. I'll spend some time this weekend to try to integrate with it. Thanks!

EricHigdon, 2020-07-09:
Putting messages in the queue is pretty simple. See line 357 of __init__.py. 

The complication is going to be in making the queue messages more universal. Right now it is either “KILL” or a progress percentage to update the progress effect. I think we need to make it into a JSON string that contains keys for progress, and brightness.

JohnTRoth, 2020-07-11:
Putting them in isn't tough, and using JSON to hold parameters wasn't bad - but I'm not too sure what triggers the queue to be processed. That, or I have some stupid error holding things back. I converted your progress functionality without issue, but when I add my stuff to the queue it never takes effect (though the first non-queued one continues to work). I know there's plenty of room for improvement in the way I have the code executing at the moment - was focusing on making it work, then I'll streamline it all.
[Comparison to my previous](https://github.com/JohnTRoth/OctoPrint-RGB_status/pull/2/commits/e0d13d241be1057854f38724ea6abd1cf6718536)

JohnTRoth, 2020-07-12:
Got everything working, but I'm not a developer so there's plenty of room for optimization. I think it would be nice if all effects blocks accepted just a json object containing all their needed parameters... but that change may be a bit beyond me.

Added fully functional M150, supporting the following:
R - Red value (0-255)
G or U - Green value (0-255)
B - Blue value (0-255)
P - Brightness (0-255)
D - Delay
I - LED Index (1 to # of LED's) When used in combination with Effect 9 (Solid With Brightness), specifying this will control the color/brightness of only the specified LED. Leaving this parameter out (or using a value of -1) will result in controlling all LED's
E - Effect (0-9) The effect to display. Leaving out this parameter defaults to Effect 9 (Solid With Brightness) to maintain compatibility with the official M150 code.

Effects are one of:
0 = Theater Chase
1 = Solid Color
2 = Rainbow Cycle
3 = Rainbow
4 = Color Wipe
5 = Pulse
6 = Knight Rider
7 = Plasma
8 = Theater Chase Rainbow
9 = Color With Brightness