EricHigdon/OctoPrint-RGB_status #16: LEDs stay white most of the time

RootUser3885, 2019-03-20:
The LEDs stay white untill the server is used or reloaded. No matter what colors set, it seems to blip the idle effect.

[image]

FlakJAG, 2019-03-20:
Mine do the same. They run constantly white at max brightness which makes them quite hot. They only show the selected color/animation for a second or two when you change settings or restart the pi. Running 16 WS2812B's on a Pi 3b.

EricHigdon, 2019-03-20:
@RootUser3885 @FlakJAG Can you please share you octoprint.log files? 

Also, do you have any other information regarding this issue? Does it happen as soon as the server loads, or after a print? If you restart the server (or change the idle effect), does it start working again?

FlakJAG, 2019-03-21:
[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/2990884/octoprint.log)
Tried different animations/ solid colors and they all do the same thing. on reboot it shows the idle color for a bit. The start up color flashes on and then white 2 or 3 times. By that point Octorprint is fully loaded and it goes to the idle color again for 5-10 seconds and then just goes all white and stays that way. If you click anything in Octoprint the idle color comes on for a few seconds before going steady white again. When I start a print it goes to the base color while warming up the printer, then goes back to full white as soon as the print actually starts. Sorry for such a rough explanation, I'll try to video ASAP to give you a better idea.

RootUser3885, 2019-03-21:
It seems to only show color when the server is loading something, otherwise they stay white at 100%. It also does it on server startup but only for a fraction of a second.
Any other information you need just ask, this is a really cool idea and i cant wait to see it 100% working.

[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/2992602/octoprint.log)
[image]
[image]

EricHigdon, 2019-03-21:
@RootUser3885 @FlakJAG I'm curious if you are both using the WS2812_STRIP LED Type?

Maybe that's the common factor that is having issues. Have you tried it with one of the WS2811 options selected?

RootUser3885, 2019-03-21:
I don't have any WS2811 strips handy, It very well could be the issue. I think my friend might have some 11s. I will have to barrow some and try it. I wonder what makes them different?

EricHigdon, 2019-03-21:
@RootUser3885 Can you try one of the 2811 options using your 2812 LEDs?

RootUser3885, 2019-03-21:
I'll give it a try, it's printing rn so it will be a while.

FlakJAG, 2019-03-21:
I tried a few different options the other day and specifically tried 2811grb again today, but I get the same results as before. I will order up some different LED's (2811's/6812's) for testing purposes. Also thought I should mention I'm running octopi_version: 0.14.0, not the latest version if it makes a difference or not.

redeye86, 2019-03-29:
Hello!

I think I have the same issue. A few days back it worked flawlessly, but now i have exactly the same problems, also on ws2812b. At first i thought it had something to do with my self soldered gpio adapter board, which i tried out the first time yesterday. But to isolate problems i tried [ws2812-spi](https://github.com/joosteto/ws2812-spi) and it works without problems. I even tried to run octoprint and wave.py from that project in parallel and the wave was interuppted by short (milliseconds) white flashes coming from octoprint/the plugin (i assume) every second or so.

Btw: I also get the configured idle animation during the slicing process using the internal cura slicer but afterwards it turns white again.

redeye86, 2019-03-29:
I have tested a few things. First i added some logging in the rpi_ws281x.py to see the color values that are actually sent to the strip. And indeed there is no 255,255,255 or so sent to them. But the logging and tail -f-ing via ssh itself was a bit cpu demanding and the solid color setting worked while the logging was active, at the same time rainbow effects still did not. 

Ultimatively i came to the conclusion that it has something to do with the timing, so i to changed the frequency setting and *tada* it seems to work fine with doubled frequency (1600000 Hz). But i do not know if this has some negative impact on something.

Btw. I forgot to mention that i also drop-in-replaced my rpi2 by a rpi3 a few days ago.

EricHigdon, 2019-03-29:
@redeye86 thanks for the info. I could totally see a setup with a lot of leds and/or other plugins running having timing issues. Just to confirm: did you add “core_freq=250” to your boot/config.txt file?

redeye86, 2019-03-29:
Sorry, you are right. I saw that hint in the instructions but when i configured my raspi 2 it was not relevant and i just didn't remember when i switched to rp3. At least it could give the others an idea what they can change. 

Interestingly the ws2812-spi library does work without that setting, so maybe it is possible to make it work without that kernel parameter.

EricHigdon, 2019-03-29:
@redeye86 So if you set the core_freq to 250, then set the frequency back to 80000Hz, do the LEDs work?

redeye86, 2019-03-29:
Yes, exactly.

EricHigdon, 2019-03-29:
@redeye86 Awesome! That's good to hear. I'm slightly ashamed that I didn't think of that when this issue first came up. Thanks so much for your help in debugging this issue. Hopefully it will work for @RootUser3885 and @FlakJAG as well.

FlakJAG, 2019-03-29:
Not having core_freq=250 was the problem, its working now. When I first set everything up I swear I added core_freq=250 to my boot/config, but I think I screwed up saving my changes. Wish I would of double checked that sooner. Thanks for the help guys. Looking forward to seeing this plugin expand, it has a lot of potential.

closed 2019-04-08 by EricHigdon

Tadimsky, 2019-04-15:
Just wanted to let you know that I had the same issue and after setting my `core_freq` to 250 on a Pi 3, it worked for me as well. 

Love the plugin by the way, I was about to build something like it a few months ago but got distracted, now when I looked again you had one :D.

Gifford47, 2020-02-05:
the same problem appears to occur again with rpi4. i have now set core_freq=500 for me, but no change, the leds glow white all the time. any advice?

Soulofdestiny, 2020-02-21:
I had the same issue, so I want to share how I solved it. When I moved the `cpu_freq=250` out from the `[all]`section into the general one, it worked for me