EricHigdon/OctoPrint-RGB_status #21: Random Flicker

RootUser3885, 2019-04-15:
Now that I got them working with the frequency fix, the LEDs will randomly flicker. It only flashes for an instant but enough to be annoying. Sometimes random LEDs are different colors than they should be. The flicker is rather spaced out, sometimes ever 5 mins, or up to 30 mins, making it hard to catch on video.
[octoprint .log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/3078362/octoprint.log)

EricHigdon, 2019-04-16:
Interesting. Did you also set your max buffer size? See https://github.com/jgarff/rpi_ws281x#spi

JohnTRoth, 2019-05-11:
I also have this on occasion. I do have my buffer size set appropriately - but suspect it's actually interference with the cable between the RGB strip and the Pi. Moving to a better cable improved things but didn't totally eliminate the issue. Next step is to shorten the length.

EricHigdon, 2019-05-11:
@JohnTRoth, yes. Cable length can be a problem, because the LEDs work on a 5v signal, but the RaspberryPi only outputs a 3.3v signal. A shorter cable and/or a level shifter is probably necessary.

closed 2019-08-06 by EricHigdon