EricHigdon/OctoPrint-RGB_status #23: cant get it to work

Kolk123, 2019-06-10:
hello.

if got a question. if installed the plugin, and enables spi.
i connect my ws2812b strip to the pi with port giopin 10.

what do i need to do more ? they dont work atm

Greetz Roy

EricHigdon, 2019-06-10:
If you've followed the directions [here](https://github.com/EricHigdon/OctoPrint-RGB_status/#running-without-root) then I would confirm that the wiring is all correct.

- [ ] Is the data line connected from GPIO10 (physical pin 19) on the Raspberry Pi to the data in of your LED strip (direction matters)?
- [ ] Is the ground pin of the LEDs connected to the ground of the power supply?
- [ ] How long is the wire between the pi and the LEDs? RPi runs on 3.3v, and most LEDs required at least 5v data. Most strips will work, but some will require a level-shifter, especially if the wire between them is too long.

If all of those things are correct, and it's still not working please share your [octoprint.log](https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299) file

Kolk123, 2019-06-11:
hey Eric,

Thank you for the responce.
i can confirm that im using the right pin. the 19e pin gpio10.
if connected the ground pin to the powersupply and the arduino.
the wiring between the rpi and the leds are max 20cm. 
if try'ed multiple led strips. when im testing it in my arduino led stair it works.

here is my log file.
[octoprint (7).log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/3275078/octoprint.7.log)

greetz, Roy

EricHigdon, 2019-06-11:
An Arduino is unnecessary. Try connecting the leds directly to the RPi.

Kolk123, 2019-06-11:
If you read my comment I dont used a arduino. I've tested it if my LEDs are working

Kolk123, 2019-06-11:
The LEDs are connected direct to the raspberry. It looks like the raspberry dont give any output on the gpio10.

EricHigdon, 2019-06-11:
Ok. Your log looks fine. 

Have you set your spiders.bufsiz and core_freq?

Kolk123, 2019-06-11:
Yes. I set it just like the tutorials. Can it be related to the newest octoprint update ?

EricHigdon, 2019-06-11:
Perhaps, though I would think that would cause errors in the log.

Just to confirm, you've restarted your pi since changing those settings and enabling SPI?

Also, which type of LEDs are you using? #22 seems to describe issues with some LED types

Kolk123, 2019-06-11:
Hey Eric. 

Thanks for responding! 
Yes I've restart multiple times.
I enabled it in the config. And checked it in the log.  I'm using 5v ws2812b leds. Maybe is that the problem.

I dont have other leds available for testing.

Greetz,  Roy kolk

Kolk123, 2019-06-11:
More detail. I'm using the rgb version. 
https://www.aliexpress.com/item/32808513592.html?spm=a2g0s.9042311.0.0.27424c4dhVVlnP

EricHigdon, 2019-06-12:
Interesting. Those look very similar to the ones that I use. Maybe they aren’t working on 3v, and you need a level shifter.

Can you share a screenshot of your rgb status settings in OctoPrint?

Kolk123, 2019-06-12:
no they working with 5v.
but i got the solution. i've swaped to another raspberry and this one works fine.
i thing one of the pie's are broken. thanks for you help :)

EricHigdon, 2019-06-12:
Awesome. Glad it’s working.

closed 2019-06-12 by EricHigdon

NickOne81, 2020-01-30:
Hi there,
I also connected the WS2812 to the Raspberry with an additional power supply, but unfortunately nothing happens. But when I run it via Python script, it works fine. I connected it to the GPIO 18 and also set it up under the Octoprint plugin. Could someone help me with that?

EricHigdon, 2020-01-30:
@NickOne81 GPIO 18 requires root access to control. You can either run OctoPrint as root, which is strongly discouraged, or use GPIO 10.

NickOne81, 2020-01-30:
Hi EricHigdon, yes, I also just read. I now have it on GPIO 10. Unfortunately it still doesn't work. I also activated SPI and buffer size and core frequency also fit. What could not fit there?

EricHigdon, 2020-01-30:
@NickOne81 does the python script that worked with GPIO 18 work with GPIO 10 now?

NickOne81, 2020-01-30:
No, unfortunately not. If I change the script to GPIO 10 it no longer works.

EricHigdon, 2020-01-30:
Are you on a Raspberry pi 4?

NickOne81, 2020-01-30:
No. I have a Raspberry Pi 3 Model B.

EricHigdon, 2020-01-30:
Did you use the wizard to change the buffer and core_freq settings?

NickOne81, 2020-01-30:
Yes, I made these settings using the wizard. The values ​​are also correct in the cmdline and in the config.

EricHigdon, 2020-01-30:
Interesting. Try adding enable_uart=1 to config.txt.

NickOne81, 2020-01-30:
Now I have found out what it is. An LCD display is connected to my Raspberry. If I take out the values ​​in the config for the display then the LED strip works.

dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900

EricHigdon, 2020-01-30:
That would explain it. The LCD probably uses the SPI pin.

I think your option moving forward from here the. Is to go back to pin 18 and run OctoPrint as root. I did confirm that it worked in the past, but be careful. https://community.octoprint.org/t/starting-octoprint-with-sudo-as-root-tells-me-you-should-not-run-octoprint-as-root-why/230