EricHigdon/OctoPrint-RGB_status #24: WS2812B WS2812 not working

CubeC62, 2019-07-09:
Hi I'm trying to get the WS2812 with my Raspberry pi 3B+ working. But my Raspberry pi 3B+ preferes to respond with the following answer (from the octoprint.log):

2019-07-09 16:17:02,785 - octoprint.plugins.rgb_status - ERROR - ws2811_init failed with code -11 (Selected GPIO not possible)
2019-07-09 16:17:02,788 - octoprint.plugins.rgb_status - ERROR - Error running effect: The strip object does not exist. Did it fail to initialize?
2019-07-09 16:17:02,796 - octoprint.plugins.rgb_status - INFO - Starting Idle Effect
2019-07-09 16:17:02,802 - octoprint.plugins.rgb_status - ERROR - Error running effect: The strip object does not exist. Did it fail to initialize?

Points I've done:
-enabled SPI
-connected the data pin from the LED strip to pin number 10 (BCM)
-connected the ground of the LED strip with the Ground of the Raspberry pi
-the LEDs are powered by a seperate PSU

closed 2019-08-31 by KSALPJan

Piranit, 2020-04-19:
@KSALPJan did you solve problem?

Piranit, 2020-04-25:
I've figured out what is the problem.
It's because the Octoprint doesn't run for sudo.
I don't suggest doing this.
Here's why: https://community.octoprint.org/t/starting-octoprint-with-sudo-as-root-tells-me-you-should-not-run-octoprint-as-root-why/230
You'd better use GPIO10 and control via SPI.