EricHigdon/OctoPrint-RGB_status #27: Can you add support for a WS2812B 5050 RGB 24 led ring?

cellfreak, 2019-08-06:
Here is the link below. 

https://www.ebay.com/itm/WS2812B-5050-RGB-LED-Ring-24Bit-RGB-LED-Integrated-Drivers-For-Arduino/201748593347?hash=item2ef9273ac3:g:gBwAAOSw241YU2w4&rmvSB=true

EricHigdon, 2019-08-06:
@cellfreak if the article you linked controls those leds, then so should this plugin. They both use the rpi_ws281x library.

cellfreak, 2019-08-07:
Your plugin does not work. I have the correct pins plugged in but the LED neopixel does not light up.

EricHigdon, 2019-08-07:
Ah, and it works with the tutorial you linked?

cellfreak, 2019-08-07:
I got it to work. But had to change the 1 to a 0 for pin 10. My PI3 started to drop the wifi connection. Thie RGB status was the only plugin I installed. I flashed a new Octoprint image with no plugins and the wifi stays connected.

EricHigdon, 2019-08-07:
@cellfreak If you mean that you changed the led_channel to 0, that is correct. In order to use pin 10 for RGB LED control, you should use led_channel 0. 

As for WiFi issues, I highly doubt that this plugin is related. It has nothing to do with WiFi.

There are still a lot of thing documentation wise that I need to improve with this plugin. It's a slow process right now with other job/life things pulling me away.

So glad to hear you got it working. Enjoy!

closed 2019-08-07 by EricHigdon