EricHigdon/OctoPrint-RGB_status #28: Adding common LED strips

tvixen, 2019-08-18:
Hi Eric
I saw your module introduction, on the Octoprint discussion group, and I think I have a idea for you. As this would probably be easy for you to implement.

Would it be possible for you to add 3 single color output (RGB) to 3 pins on the Pi, so every color has it's own pin ? (on/off)  In this way the status indication would work with LED strips too.

I know this code is for ws281x SPI, but I think that adding this, the people could choose between common LED strips and Neopixel LED strips :D  (not in so many colors as neopixel, but 7 is also ok)

The only thing people has to do, is to add 3 transistors, and 3 resistors to drive the 3 colors of the LED strip. And then connect it the gate to pi, the LED to drain and the source to the power supply.

Br Tim

EricHigdon, 2019-08-18:
I thought about this. It would be a cool feature. PWM support would allow full color control (not just 7). 

This is not going to be a high priority for me when I get back to this project though because there are other options for controlling regular LED strips. https://plugins.octoprint.org/plugins/ledstripcontrol/

tvixen, 2019-08-18:
Yes I saw that. 
And, I did try it, but it doesn't seem to work with Marlin 2.0.x
I can send M150 code from pi to it (it works), but enabling the #define RGB_LED in the Marlin software, is not doing anything for the print event.

yorb, 2020-06-08:
Adding a vote for this. As mentioned, none of the common LED strip control plugins allow control based on printer events like this one does. It would be a great feature!

CraXyOW3, 2020-09-12:
I second @yorb 's comment. 👍

Titi9237, 2020-11-15:
I third @yorb 's comment. +1