EricHigdon/OctoPrint-RGB_status #29: Neopixel and Level Shifting Chip

kzmodelno, 2019-08-21:
Hello! 

First time poster here.  I am new to Neopixels and the RPI in general, but I believe I have everything wired and installed correctly.  I am using a long strip of LED's (53 in total) in conjunction with this OctoPrint plugin and I cannot get the LED's to reliably produce color.  They will turn on and stay white, occasionally throwing color for a few seconds.  The colors seem to coincide with the "ACT" light on the RPI.  I wired the lights using the first diagram here with the 74AHCT125 level shifting chip: https://learn.adafruit.com/neopixels-on-raspberry-pi/raspberry-pi-wiring

My question is thus: Is there any change required due to the level shifting chip?  Could the frequency of the signal be affected by using the chip?

EricHigdon, 2019-08-21:
@kzmodelno welcome. 

You shouldn’t have to do anything different if using a level shifter. Have you followed all of the steps in the running without root section of the readme? https://github.com/EricHigdon/OctoPrint-RGB_status/blob/master/README.md#running-without-root

kzmodelno, 2019-08-22:
Hi Eric,

Thanks for the quick reply! After following the detailed SPI section, things are working as expected.  I appreciate your time and the great plugin.  Thanks!!

closed 2019-08-22 by kzmodelno