EricHigdon/OctoPrint-RGB_status #30: Add Raspberry Pi 4 support

EricHigdon, 2019-09-07:
rpi-ws281x 4.2.0 supports the Raspberry Pi 4, I'll need to upgrade that dependency, then test everything on a 3 and a 4.

joeynuggetz, 2019-10-19:
So should one expect that the RGB status doesn't work on a pi4? I've been trying all night to get it to work. Same connections/settings as pi3 and on Pi3 the led ring works flawlessly but I get nothing on my pi4.

EricHigdon, 2019-10-19:
I have a rpi4, but haven’t had a chance to install OctoPi on it yet. You could try manually upgrading the rpi-ws281x package on your machine. It may work out ok.

joeynuggetz, 2019-10-19:
Not sure what I just did but I tried to install rpi-ws281x package and it seems to be working outside of octoprint. How do I get RGB status to use whatever it needs to function? Can I update rpi-ws281x somehow within octoprint?

Timwun, 2019-11-20:
Same here with a RPi 4.

Get `octoprint.plugins.rgb_status - ERROR - ws2811_init failed with code -3 (Hardware revision is not supported)` in the logs.

devildant, 2019-11-20:
it's a problem with the SPI, when running in an independent script on the GPIO 18 there is no problem.
but as soon as the SPI (gpio 10) is used, the led switch between color and white between each call.
rpi_ws281x 4.2.2

Kedryn, 2019-11-25:
I've an RPI 4 with octoprint and Enclosure. Testing a 16led circular WS2812,  on GPIO14 and GPIO18, and isn't working. All i get is the first led->green, all others->off

redeye86, 2019-12-05:
I had the same white-with-occasional-colorful-flashing-issue when i changed from a 2 to a 3. As I just put the sd card to the new pi, i didn't read the installation instructions of rpi_ws281 which mention that you have to change the core frequency (see https://github.com/jgarff/rpi_ws281x#spi). I guess you have to adjust them again for the pi 4.

derivmug, 2019-12-30:
I am also facing this issue. I could maybe have a look into this however I haven't developed any OctoPrint plugins before. Over at the rpi-ws281x repo there seems to be a temporary solution: [](https://github.com/jgarff/rpi_ws281x/issues/384). I might try it in the following days.

EricHigdon, 2019-12-30:
@derivmug thanks. My time and LED resources are currently devoted to my Christmas display. I will shift focus back to this plugin soon, hopefully.

dexer12, 2020-01-19:
Hope you enjoyed christmas and coming good to 2020.
I tried to change the dependency on my own, but just changing the required version number wont fix it at all :( but i'm not too much into this to get it working for me. Spend already a few days without success :(
So i hope you'll find some time to get it up running on the pi4.

EricHigdon, 2020-01-21:
I upgraded the required version of rpi-ws281x to 4.2.3 (latest) which is supposed to include support for the Raspberry Pi 4. Have not tried finished installing OctoPi on my rpi4 yet, but an initial look seems like the problem reported by some here may be reported here: https://github.com/jgarff/rpi_ws281x/issues/381

EricHigdon, 2020-01-21:
Once I got my new rpi4 up and running with OctoPi, everything is working fine (with the updated rpi-ws281x dependency). [Version 1.6.0](https://github.com/EricHigdon/OctoPrint-RGB_status/releases/tag/V1.6.0) officially supports RaspberryPi 4B.

Anyone still experiencing issues, please double check the [Running Without Root](https://github.com/EricHigdon/OctoPrint-RGB_status#running-without-root) section of the readme, or uninstall then reinstall the plugin and use the setup wizard.

closed 2020-01-21 by EricHigdon