EricHigdon/OctoPrint-RGB_status #32: Setup Wizard appears every time i visit the Octoprint page

cs2000, 2019-09-23:
Not sure if this one is just me or... but since upgrading recently to the new RGB Status version that has the rather nice setup wizard, i cant seem to get it to go away!

I performed all the steps it wanted me to, and have green ticks by them all, but every time i access the Octoprint admin page, the setup wizard comes back up again. Its still all green ticks, i just have to press finish and it goes away, 

Its just rather annoying it does this at every login.

EricHigdon, 2019-09-23:
Interesting. OctoPrint is supposed to be handling the hiding of the wizard if you’ve already completed it, I’ll have a look.

cs2000, 2019-09-23:
Appreciate it, if you need any logs etc, let me know, happy to provide.

EricHigdon, 2019-09-23:
I think this is a bug in the current version of OctoPrint (1.3.11). It will be fixed in 1.3.12, but until then, you should be able to fix it by holding shift while clicking the refresh button in your browser. 
https://github.com/foosel/OctoPrint/issues/3211

closed 2019-09-23 by EricHigdon