EricHigdon/OctoPrint-RGB_status #33: Reverse LED Direction & More

cs2000, 2019-09-26:
Not an issue, more of a suggestion, or two!

1. Provide an option for the “print status” lighting mode to reverse the direction of the progress bar.

In my setup, the strip wires come in on the right hand side, meaning the progress changes from right to left rather than left to right.

I know that for some people, just swapping the led tapes mounting direction may be easy to do, in some cases however this may not be possible due to access or wiring.

—- 

2. Timeout of some LED states.

For example, if a print fails, or actually if I click Cancel in octoprint, the LED’s pulse red, which is great. However, my RPI always stays powered up. So I may turn the printer totally off and give up for that day, but the led tape is still flashing red.

The only way I get get them to go back to the “idle” state, is to reboot the RPI, or change a setting in the RGB Status options, then they seem to go back to “idle” (which in my case is off)

—-

Other than that, an amazingly useful plugin, that especially with the new setup wizard, actually works really really well! Makes it super simple to see my print status just by a quick glance at the led’s. Top work!

EricHigdon, 2019-09-29:
These are some good points. I’ll consider these for v. 2.0. 

Thanks for the encouragement and suggestions!

drewzh, 2020-01-22:
Hi @EricHigdon, I just bumped into the same issue with the LED direction. Have you had any time to work on this? Thanks

EricHigdon, 2020-01-23:
1. is handled in 4ec7682

2. will be covered by #37

closed 2020-01-23 by EricHigdon