EricHigdon/OctoPrint-RGB_status #38: Two WS2812 Strips...

Snille, 2019-11-02:
Hi, I'm trying to set up two WS2812 strips (for two instances of Octoprint) on a Raspberry Pi 3+. The first one uses SPI0 (BMC 10 and works), the other one I would like to add to SPI1 (BMC 20).

I'm running Octopi (so user pi is in the GPIO-Group). Both SPI interfaces are enabled. I can see them in /dev

pi@octopi:~ $ ll /dev | grep spi
crw-rw----  1 root spi     153,   0 Nov  2 13:12 spidev0.0
crw-rw----  1 root spi     153,   1 Nov  2 13:12 spidev0.1
crw-rw----  1 root spi     153,   4 Nov  2 13:12 spidev1.0
crw-rw----  1 root spi     153,   3 Nov  2 13:12 spidev1.1
crw-rw----  1 root spi     153,   2 Nov  2 13:12 spidev1.2

But only the first strip works... What have I missed?

EricHigdon, 2019-11-03:
Have you had a look at the settings for this plugin?

Am I understanding properly that you are running two instances of OctoPrint on the same RaspberryPi? If so, you’ll want to make sure one of them is set to pin 10, while the other is set to pin 20.

That said, I’ve never tried running two instances on the same machine before and not sure if it will play nicely with the SPI control.

Snille, 2019-11-03:
Hi, yes, you are correct. I'm running [two instances](https://www.youtube.com/watch?v=7Saa1HpLRoM) of Octoprint on the same PI, you can do that without any problem. Each instance gets it's own settings. I have set the pins to 10 (in the first instance and 20 in the other... :) However, I'll try the PWM-path. Thank you, I'll let you know if it works when I have tested.

Looking forward to 2.0! :)