EricHigdon/OctoPrint-RGB_status #42: Configuration/pins

Grk093, 2019-11-18:
Hi,

First of all, thanks for the amazing plug in, it should be of one "must have" for Octoprint. 
I'm would like to ask, a lame question, but how the led strip should be connected to raspi? I mean, I have WS2812B LED strip with five wires. I would like to connect it to raspi 4 (3, 4 or 5 LEDs). How the wiring should be done? 

I should cut the plug and connect it like this? 
[https://tutorials-raspberrypi.de/wp-content/uploads/Raspberry-Pi-WS2812-Steckplatine-600x361.png]()

Or I must to use mosfets?
[http://lucylimd.com/a/a/co/controllable-christmas-lights-using-raspberry-pi-rgb-led-strip-aruls_mosfet-transistor_nmos-transistor-pnp-electronics-bc109-nkt275-series-pass-common-transistors-what-is-tutorial-on-amp.png]()

Maybe I should add some resistors and capacitors?
[https://i.stack.imgur.com/fXj79.png]()

Thanks in advance!

EricHigdon, 2019-11-18:
If you truly have ws2812b LEDs, then you should have only 3 wires (vcc, gnd, and data in). I’m which case, the first image you linked is correct.

Grk093, 2019-11-18:
Thanks for the answer. This means, I have to cut the plug. 

I have these ones: [https://www.aliexpress.com/item/32337440906.html?](url)

EricHigdon, 2019-11-18:
Right. You could cut the plug off, or you could get another plug to attach to the RPi. Up to you.

closed 2019-11-18 by EricHigdon

mikekscholz, 2019-11-29:
the extra 2 wires are most likely just extra 5v and ground wires for connecting extra external power.