EricHigdon/OctoPrint-RGB_status #43: N channel mosfet usage

mikekscholz, 2019-11-29:
Ive been searching for days about the proper way to do this and everything I find is counter intuitive and seems like it would fry my pi if I tried it. Logically I think that +5v should be connected to drain, GPIO 10 connected to gate, and source connected to DI on the led strip. If im wrong what is the proper way to do this because buying a bidirectional logic shifter seems overkill when I only need to shift a single line in one direction and I have bags of different mosfets sitting around already.

EricHigdon, 2019-11-29:
my first suggestion is always to try it without using a level shifter. The 3.3v has always been enough for me to drive my LEDS unless they were > 15’ from my RaspberryPi.

Checking if 3.3v input works for something that is expecting 5v shouldn’t really have a chance of burning anything up, so it’s always worth a shot.

A mosfet on the data line *might* work, and the connections you describe seem correct.

mikekscholz, 2019-11-29:
thats my only experience with them so far as well. Im not getting any activity on my strip so I connected my leads to a itsybitsy m4 that has a 5v output to make sure I didnt damage any of my wiring and they work fine, and I double checked that I made all of the changes in the spi section of the ws281x lib page so the output voltage is the only thing I can assume is still not correct and my wiring is only about 18". when i was doing oldschool rgbs is was gate to gpio, drain to led and source to ground but that was switching the ground instead of the positive.

EricHigdon, 2019-11-29:
Ah. That background info is helpful. Now that I’m thinking about it, I’m not sure if a mosfet will work the other way. Are you getting any errors in your OctoPrint.log?

mikekscholz, 2019-12-01:
Never thought to check the logs, never had a reason to yet. My SPI was failing to initialize and no matter what I couldnt get it started. I flashed a fresh copy of octopi onto one of my rotating backup cards and did the same config over again and the lights fired up as soon as I installed the plugin without even needing to configure any other settings

closed 2019-12-01 by mikekscholz