EricHigdon/OctoPrint-RGB_status #46: [Feature Request] Add Manual On/Off toggle for LEDs

j0shm1lls, 2019-12-19:
Love the plugin!  Just hoping to get more control by perhaps adding a manual on/off toggle to the sidebar or perhaps on the control page.

Thanks!

Wummeke, 2019-12-31:
I just was about to post this request too :)

Scope666, 2020-01-22:
This would be AWESOME.  This plug-in rocks already, and has saved me from repeatably fighting with Marlin / BTT's horrible Neopixel fork... this just works!

drewzh, 2020-01-22:
I think expanding this to a few utility functions would be very handy for testing effects and that you've got everything setup correctly. It's quite cumbersome to keep going into settings, changing idle to X and then hitting save, to then have to go through the same routine again to test another colour.

Or better yet, add a special Gcode hook to listen for gcode events (and then the custom buttons can be added as helpers in the UI separately).

Scope666, 2020-01-22:
If it could "listen" for gcode M150, or something else that we could send, we could leverage Custom Control Editor to make preset buttons.  That's what I was doing when I was directly in the Neopixel port of the SKR.

EricHigdon, 2020-01-22:
This is going to be a global on/off button that goes in the navbar. 

I like the idea of a controller UI that allows you to test effects on the fly. I created #51 to handle that though. 

I don't think M150 gcode is the right way to do this as it is only intended to support a single LED, except where RepRap kind of hacked dotstar support in. https://reprap.org/wiki/G-code#M150:_Set_LED_color

EricHigdon, 2020-01-22:
Done in c612e5f

closed 2020-01-22 by EricHigdon