EricHigdon/OctoPrint-RGB_status #48: Newly installation on RP4b

Gnizten, 2020-01-21:
Installed the latest update, that was compatible with RP4.
Could not get the plugin to work before, but now there is life.
But i experience some issues, and i dont know if this is normal behavior.

Im running the WS2812w strip on the pi, and heres what happening.

-When idle/octoprint idle, the LED strip is white.
-When printing and configured to show print status, the status is showing, but the led strip shifts between all white and the status with the green led? the white fase shouldnt be there or ?
When finish printing i have chosen the knight rider theme, but the same thing happens with the shifting between all white and the theme.

it seems that the all white fase is interfering with all states.

br Tom

EricHigdon, 2020-01-21:
@Gnizten that sounds a lot like a timing issue. Did you set the core frequency?

dexer12, 2020-01-21:
I tried the new version too. Thank you a lot to put effort into this again @EricHigdon !
But it seems to me that he is playing the pattern and after playing it once, switching back to just white. I think this should be running in a loop.

Whats curious to me, that the WS2811_STRIP_GBR is preselected, whats fine. But it seems that there is no difference in selecting the WS2812_STRIP, what i think you fit the LEDs i'm using.

I also checked that the entry for the core_freq=250 in th /boot/config.txt exists, but seems to be fine for me.

EricHigdon, 2020-01-21:
Can you share your OctoPrint.log file?

dexer12, 2020-01-21:
The hint with the core_freq was the good. I took some research and it looks like the PI4 is ignoring this setting in the /boot/config.txt. Dont ask me why.
I found an issue somewhere else, where the core_freq was locked down to 250, because there was an entry in the /boot/config.txt called enable_uart=1 .
I now added this entry and the LEDs are working like expected. So i think, now the core_freq is really 250 because the activated uart forcing the PI to this.

I added the octoprint.log as txt file. Its just a new clean setup, with just rgbstatus plugin.

[octoprint.txt](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4093740/octoprint.txt)

EricHigdon, 2020-01-21:
@dexer12 oh. That sounds really familiar. Thanks for the info.

dexer12, 2020-01-22:
I need to check this as soon as i get some time to. But is it really neccessary to set the core_freq?
As i played with plugin and the rpi_ws281x library the test script from the library seems to work fine, as i might know now, the core_freq=250 settings didn't do anything.
But i really need to check if it was working 100% or just run into the same issue.
Just trying to understand these things and behaviour.

EricHigdon, 2020-01-22:
It is necessary to set the core frequency to 250MHz. Otherwise the timing us not consistent and you get things like the white flashing described here. 

According to the [UART docs](https://www.raspberrypi.org/documentation/configuration/uart.md), setting enable_uart_1 also sets core_freq=250. So if you did that, you don’t need the explicit core_freq setting.

Gnizten, 2020-01-22:
Hi Eric. I could not answer this earlier. No i did not set the core freq.
because i thought that the installation of the plugin took care of this. I
can see in the last post, that i just need to add /boot/config.txt called
enable_uart=1, which I will try out tonight.

Are we sure that setting core freq to 250, will not slow down the
performance of Octoprint in general?

Den ons. 22. jan. 2020 kl. 13.16 skrev Eric Higdon <notifications@github.com

Gnizten, 2020-01-22:
Geniously... Now all is working as expected, as for what i have seen until
now. Thanks for now.

Br Tom

Den ons. 22. jan. 2020 kl. 18.52 skrev Tom Schou Larsen <
tlarsen@drosselvangen.dk>:

EricHigdon, 2020-01-22:
@Gnizten Awesome. Glad it's working out for you.

As for your performance question, the name `core_freq` is a bit misleading. That setting is actually changing the GPU core frequency, not the CPU core frequency (which is set with arm_freq).

Changing the GPU frequency could affect the CPU performance because it also controls the memory bus, but the difference on the pi2 and up is minor. 

That said, part of the pi4 upgrade was the better GPU and "4k Monitor Support" If you are trying to actually run octopi and high resolution graphics, you will probably have issues with setting your GPU core to half of the default of 500.

Now I just need to decide how to move forward. On the pi4, a core_freq value of 250 is ignored, because the only accepted options are 500 (default) or 600. The enable_uart=1 setting is a workaround, but I'm not sure I want to build that into the setup wizard since it is a bit hacky.

closed 2020-01-22 by EricHigdon

edsai, 2020-02-04:
I attempted the enable_uart fix and the LEDs did the rainbow startup a few times and then shut off. Without it they would do the same white flashing behavior. I do have the enclosure plugin installed before and I'm not sure if there's some dependency conflict that happens when enable_uart is configured. I noticed in the other issue you mentioned to run the setup wizard again but I'm not sure how to do that since there doesn't seem to be a web ui method to do it. Uninstalling the plugin and reinstalling did not yield any different results.

EricHigdon, 2020-02-04:
@edsai Can you please share your OctoPrint.log?

edsai, 2020-02-04:
Sure. Here you go.
[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4153852/octoprint.log)

edsai, 2020-02-04:
For some reason it's working now. I rebooted multiple times after making my settings changes to try to get it to work. I didn't do any updates but I was showing my daughter the other octoprint enclosure work I had done and all of the sudden the LEDs were working.