EricHigdon/OctoPrint-RGB_status #50: Consider Replacement for core_freq=250 Setting on Raspberry Pi 4

EricHigdon, 2020-01-22:
On the pi4, a core_freq value of 250 is ignored, because the only accepted options are 500 (default) or 600. The enable_uart=1 setting is a workaround, but I'm not sure I want to build that into the setup wizard since it is a bit hacky.

dexer12, 2020-01-23:
I already found some information, where it seems that the Pi4 is underclocking the core_freq to 200mhz when in idle mode. I could not test it yet, but i would give this a try:

core_freq=250
core_freq_min=250

As far as i understood what i've found. the core_freq=250 sets the core freq correctly but regarding the other behaviour of the Pi4 it will use a core_freq of down to 200 while being idle.
So maybe this could be fixed with setting the core_freq_min.

EricHigdon, 2020-01-23:
@dexed12, can you share links to the docs where you found this info?

dexer12, 2020-01-24:
I first found it here:
https://github.com/raspberrypi/linux/issues/3381#issuecomment-568546439

And in the documentation from raspberry here:
https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

where you could see that the core_freq changes for example when enableing 4k 60 FPS Hdmi Mode.

EricHigdon, 2020-06-13:
https://github.com/jgarff/rpi_ws281x/issues/381 Indicates that this issue can be resolved by setting

`core_freq=500
core_freq_min=500`

I’ll give this a shot soon and implement it into the wizard if it works.

Note to self: I need a way to show the wizard again to someone who has already seen it if they are on an rpi4.

cp2004, 2020-06-13:
Can confirm that this works on all my Rpi4s, you need both as it clocks down when not at high demand (Like higher end CPUs) and screws up the timing.

unexist, 2020-07-14:
I cannot confirm, that this solution works for my Ender 3 Pro/Rasp 4B and according to the docs this combination isn't supported at all:

https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

**core_freq** is ignored as stated in the docs:

`Changing core_freq in config.txt is not supported on the Pi 4, any change from the default will almost certainly cause a failure to boot.`

And the allowed values for **core_freq_min** are **250** and **275**, according to the table in the docs.

Additionally, when I use above combination, I cannot establish any connection between my Ender and my 4B.

So what works for me so far is to just use `core_freq_min=250`, although I cannot say if the progressbar works at all, my current print job is at 0%. ;) 

(Will update later, if I can see any progress tho.)