EricHigdon/OctoPrint-RGB_status #55: wipe color after octoprint shutdown

Gifford47, 2020-02-05:
the problem: after octoprint shutdown the last effect is killed but the leds remain white.
a fix would be: wipe color after shutdown event

Gifford47, 2020-02-09:
ok, i have tried several times, but i couldn't figure out why the leds don't turn off. either on direct startup or on shutdown. the problem is that the white color needs the most current and therefore the pi registers voltage problems on startup.

recently i tried the following without success:
    def on_shutdown(self):
        self.kill_effect()
	self.color_wipe()

does anyone have any idea why the color wipe is not being performed?

EricHigdon, 2020-02-09:
@Gifford47 see my comment on your pull request (#56). color_wipe is not a method on the plugin class. You should use self.run_effect() with the necessary parameters.

Gifford47, 2022-08-25:
in the latest code, the color is wiped correctly. no need for investigation anymore. thanks for feedback!

closed 2022-08-25 by Gifford47