EricHigdon/OctoPrint-RGB_status #57: WS2812 (Adafruit Neopixels) cannot work on GPIO18 (pin 12)

clevere12, 2020-02-18:
Works on GPIO10, but the LEDs do not show accurate color, however when tested using the Adafruit libraries, the LEDs show accurate color just fine on GPIO18 (pin 12). 

It would be nice to be able to use PWM on GPIO 18. I know it requires root, but there seems like there could be a way to make it work. If anyone knows how ...

EricHigdon, 2020-02-18:
@clevere12 what do you mean by accurate color?

You can use PWM on GPIO 18. You just have to run OctoPrint as root, which is not recommended.

clevere12, 2020-02-18:
Color accurate meaning, on GPIO18, Red is red, green is green, orange is orange, etc. Using GPIO10, Red is more of a pink, Orange is some off-white color, blue is more of a light blue.

Yes, I don't want to run it as root, but this octoprint is running off. the octopi image and not available to the outside world, so technically it could be run that way, don't want it though. Seems like there should be a work around, but hell if I can find it.

clevere12, 2020-02-19:
I wonder, could control of GPIO18 be had by adding a sudo command option to the program? I was looking at an enclosure plug in with this option... just trying to figure it out and I appreciate your help and your creation.

OscarWasa, 2020-03-20:
Hi Eric, I am new to this and I have problems with the leds w2812b, I have them connected GPIO 10 = PIN19 but I don't have to put in PIN LED and LED DMA