EricHigdon/OctoPrint-RGB_status #60: Unable to get running after updates

JohnTRoth, 2020-03-15:
I've been using this plugin for about a year, but recently upgraded the plugin and OctoPrint, then suddenly found it no longer loading at all. I was initially getting:

> ImportError: /home/pi/oprint/local/lib/python2.7/site-packages/_rpi_ws281x.so: cannot open shared object file: No such file or directory

So after some searching, recompiled and reinstalled the rpi_ws281x library. That got it to the point where the plugin will load, but fails to initialize with:

> octoprint.plugins.rgb_status - ERROR - in method 'ws2811_channel_t_gamma_set', argument 2 of type 'uint8_t *'

I'm able to run /home/pi/rpi_ws281x/python/examples/lowlevel.py under both python 2 and 3 without issue as the pi user, so I'm quite confused as to where my issue is.

I'm fairly sure I don't have settings issues as it has worked previously, but for reference, I'm using this in the lowlevel.py test:
```
LED_CHANNEL    = 0
LED_COUNT      = 16         # How many LEDs to light.
LED_FREQ_HZ    = 800000     # Frequency of the LED signal.  Should be 800khz or 400khz.
LED_DMA_NUM    = 10         # DMA channel to use, can be 0-14.
LED_GPIO       = 10         # GPIO connected to the LED signal line.  Must support PWM!
LED_BRIGHTNESS = 255        # Set to 0 for darkest and 255 for brightest
LED_INVERT     = 0          # Set to 1 to invert the LED signal, good if using NPN
                                                        # transistor as a 3.3V->5V level converter.  Keep at 0
                                                        # for a normal/non-inverted signal.
```

And for my settings in OctoPorint, I'm using:
```
LED Type: WS2812_Strip
LED Count: 16
LED Pin: 10
LED Frequency: 800000
LED DMA: 10
LED Channel: 0
```

Any help would be appreciated!

EricHigdon, 2020-06-13:
This is weird, it seems like something is wrong with your rpi_ws281x installation. Are you using python2 or 3 for OctoPrint?

JohnTRoth, 2020-07-03:
I think I had done all sorts of things independant of the plugin, and recently got a touchscreen as well, so I formatted and started over. Now have it running (properly) along with the 3.5" touchscreen and all is well... must have been something I did horribly wrong. Never mind!

closed 2020-07-03 by JohnTRoth