EricHigdon/OctoPrint-RGB_status #66: [Question]Is it bad practice to power LED from the 5v pin(s) on the Raspberry PI?

tomohulk, 2020-05-23:
Im relatively new to these low voltage electronics.  I am currently running an Ender 3, with a buck converter, spliced onto the PSU, converting the 24v to 5v and powering my Raspberry Pi 3B via USB from the converter, it is running Klipper and Octoprint.  I recently bought a [WS2812 24 LED ring](https://www.amazon.com/gp/product/B07YZ4H5DB/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1&tag=tv-auto-20), and have been designing this [remix](https://www.thingiverse.com/thing:4376236) to work with the smaller ring/Bullseye Duct.

It "seems" to work, im still in the prototyping phase.  Im just wondering if its good/bad practice to power this LED ring from the GPIO pins from the PI, or, should I be running the VIN and GND right into the buck converter?

great project by the way, really love the concept!

cp2004, 2020-06-07:
It's generally _very_ bad practice to run LEDs straight from the GPIO for anything above a handful. The pi can only provide any leftover power - The Pi 3B + has a 2.6A Fuse (If my googling is correct) so you get `2.6A - Pi power consumption - peripherals` and that includes USB so if you have a camera that draws from that, as well as some printer boards can draw from the USB if their main power is out (Shouldn't be an issue in your case)

It may work in testing, but who knows when the Pi's CPU decides it needs more power. Much safer to power straight from the printer so you skip the chance of shutting off the Pi for too much current draw.

EricHigdon, 2020-06-07:
That JS for covering this @cp2004. Agreed that pulling all of your power from the pi pins is not the best idea.

This doesn’t mean that you can’t use the same power supply for both the pi and your LEDs though. Just tap into the PSU for both the RaspberryPi and the LEDs.

I would personally use two separate buck converters, but you could probably get away with just one.

closed 2020-06-07 by EricHigdon