EricHigdon/OctoPrint-RGB_status #70: Toggle switch not functioning

orgalinq, 2020-06-13:
Upgraded today from 1.6 to 1.7. After that the plugin in not working anymore. The toggle switch in the nav-bar visually switches on and off on the screen but not the lights.

Using PI 3b+

EricHigdon, 2020-06-13:
Can you share your OctoPrint.log please?

orgalinq, 2020-06-13:
Here it is:

  OrgaLinQ

kokkelmanlanden 23
7542 JC  Enschede
+31(0)651 608 840

(IPhone bericht)

EricHigdon, 2020-06-13:
@orgalinq the file is not attached. 

I don’t think GitHub supports adding files from email attachments if that’s what you are trying to do.

orgalinq, 2020-06-13:
Ok, here is the octoprint.log

Thank you.
[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4774741/octoprint.log)

NameOfTheDragon, 2020-06-13:
I can confirm this. On mine, the effect freezes but the lights don't turn off. 

Also, I noticed that saving the settings sometimes freezes the effect until the lights are toggled off and on from the navbar. However in the process of making this comment I noticed that I had my type of LED set incorrectly, and after I changed it I could no longer reproduce the freeze-after-settings-change thing. The lights still don't turn off though.

I'm using a strip of 7 neopixels, part ADA1460, LED type SK6812IC. Here's my config screen in case it helps...
[image]

EricHigdon, 2020-06-13:
@NameOfTheDragon Can you share your OctoPrint.log as well please?

NameOfTheDragon, 2020-06-13:
[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4775098/octoprint.log)
Here's my log.

I've noticed an additional related issue. If the lights are toggled off then on during a print, they resume in the incorrect status (I get idle state, rainbow wipe, instead of printing state, all white).

cp2004, 2020-06-13:
@NameOfTheDragon Does it resume the printing state once the progress bar changes again? That's how it's working my end. 
@EricHigdon Maybe want to save a 'last effect' state so that on lights on you can just go straight back into it?

cp2004, 2020-06-13:
For the main issue, for me it works, but is not an instant response. (Couple seconds delay) Could be due to the time it takes for effects to exit, I will have a look at if there are any waits that could be sped up

EricHigdon, 2020-06-14:
@orgalinq @NameOfTheDragon This issue should be fixed in V1.7.1.
Please update and let me know if you still have this problem.

closed 2020-06-14 by EricHigdon

NameOfTheDragon, 2020-06-14:
I'm not sure if this is an odd Octoprint thing, but when installing the 1.7.1 update, after restarting Octoprint is prompting me to install it again. The plugin manager says I've still got 1.7.0 installed. When I try to reinstall it, in the installer output it says something like "This version was already installed, forcing a re-install". The fix IS working though, I can now toggle the lights on and off.

EricHigdon, 2020-06-14:
@NameOfTheDragon Try re-installing one last time please. I just forgot to change the version number in the plugin code.

orgalinq, 2020-06-14:
Have upgraded from 1.7.0 to 1.7.1. All right now, the LEDs can now switch on and off. At first glance everything else works ok too.
Happy with it, thank you!