EricHigdon/OctoPrint-RGB_status #71: 1.7.0 Thoughts

cp2004, 2020-06-13:
Not really an 'issue' as such, but as you said in #65 I'll give you my findings from my initial testing:
- It works 👍
- Restarting is yet to be tested (Give me 10 mins)
- The strip is generally slow to respond: Could just be waiting for effect process to spin up/down. I have a few ideas that I was working on before you released 1.7 to fix the issues, and at the moment I think it responds faster. If I can finish it I'll ping you so you can have a look
Looking through the code, it is getting more complicated! I was thinking it could be a little simpler but if it works...

cp2004, 2020-06-13:
The only difference is that the printer is running Python 2, and my dev Python 3. I'll test if it is Python 3 vs 2 in a minute

EricHigdon, 2020-06-13:
Ok, so something is up with python 2 now. I thoroughly tested both versions. The only thing I can think of is that maybe running python 2.7 on startup causes an issue. I’m currently provisioning an additional RaspberryPi with a fresh OctoPrint install to try to track down the issue.

cp2004, 2020-06-13:
I've just managed to get it to work. Nothing was showing in logs for RGB_Status for python 2 but after a couple of restarts on the dev pi it worked. No idea!

cp2004, 2020-06-14:
@EricHigdon thank you so much for your work on this, just updated the Printer to 1.7.1 and it worked beautifully. Ping me if you need any more testing or help for new features!

closed 2020-06-14 by cp2004