EricHigdon/OctoPrint-RGB_status #72: Can't upgrade from 1.6.0 to 1.7.1

orgalinq, 2020-06-16:
Hello,
Yesterday successfully upgraded from 1.6.0 to 1.7.1.
Today, on another PI & printer, I can't upgrade.
I get error codes, see my Logs. 
Can you help me to fix this?

[RGB install error logs.txt](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4789101/RGB.install.error.logs.txt)

Thank You very much in advance.

cs2000, 2020-06-17:
I too am getting this issue, even tried removing the plugin, rebooting and reinstalling, same error messages as the above commenter.

`Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-2ZACPx/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

wees, 2020-06-19:
Also having the same error when trying to update:

`Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-y0tfxp/`

EricHigdon, 2020-06-20:
Can someone try updating again? I think I have this issue resolved. in V1.7.2.

cs2000, 2020-06-20:
Worked first time for me this time around. Thanks for the fix!

closed 2020-06-20 by EricHigdon