EricHigdon/OctoPrint-RGB_status #77: RGB_STATUS staying at 1.7.1 even though says update to 1.7.2 successfull

WilkBoy, 2020-06-29:
I have tried updated to 1.7.2 from 1.7.1 a couple of times now. OctoPrint reports that the update was successful and reboots, but remains at 1.7.1 after the reboot.

Attached is my log:
[octoprint.log](https://github.com/EricHigdon/OctoPrint-RGB_status/files/4846229/octoprint.log)

I could try removing the plugin and adding/configuring it again. Any suggestions? Thanks!

orgalinq, 2020-06-29:
I have the same issue.

NameOfTheDragon, 2020-07-06:
I'm seeing this too. This happened before and I suspect the author just forgot to update the version coded into the plugin.

Maybe you could use something like [GitVersion](https://gitversion.readthedocs.io/en/latest/input/docs/) to automate this?

johnrbussard, 2020-07-11:
I’m experiencing this also.

orgalinq, 2020-07-12:
I’m experiencing this also.

Gianlucorso, 2020-07-25:
Hello,
I had the same issue in these days.
I have solved it by unistalling it through the `Plugin Manager` into Octoprint web interface, and reinstalling it by command line.
Once disinstalled, I connected through SSH to the Raspberry connected to the printer, then I typed the following command from the `home`:
```
$ ./oprint/bin/pip install "https://github.com/EricHigdon/OctoPrint-RGB_status/archive/master.zip"
```
In this way the newer version (`1.7.2`) will be installed.

> Reference: https://plugins.octoprint.org/plugins/rgb_status/

Laess3r, 2020-07-26:
I have the same issue too. Keeps saying "1.7.1" installed successfully when updating to 1.7.2.

orgalinq, 2020-07-28:
> Hello,
> I had the same issue in these days.
> I have solved it by unistalling it through the `Plugin Manager` into Octoprint web interface, and reinstalling it by command line.
> Once disinstalled, I [...]

Thank you very much. That did the trick.