EricHigdon/OctoPrint-RGB_status #78: Not working on Pi4

clevere12, 2020-06-30:
I have this running successfully on the Pi3 B+.

Using the exact same LED setup, tested to work functionally on the 3B+, I can't get any LEDs to light up on the Pi 4B+. I uninstalled the plugin and reinstalled it, thinking I messed up putting in the password for it to make config files changes, but after it has done it's initial setup, even uninstalling doesn't give you the option to put in the password for the file changes and I can't find reference to it on the GitHub page...

Soo .. with SPI enabled and confirmed enabled, is there a way to test RGB status, physically w/o using the plugin so I can trouble shoot?

`Starting" to "Printing"
2020-06-29 23:02:17,030 - octoprint.plugins.rgb_status - INFO - Updating Progress LEDs: 0
2020-06-29 23:02:17,538 - octoprint.plugins.rgb_status - INFO - Updating Progress LEDs: 1
2020-06-29 23:02:26,426 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2020-06-29 23:02:26,435 - octoprint.util.comm - INFO - Force-sending M108 to the printer
2020-06-29 23:02:26,463 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: I_dont_know_v11_0.2mm_PLA_MK2.5SMMU2S_35m.gcode, owner: XXXXX, user: XXXXXX
2020-06-29 23:02:26,469 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-06-29 23:02:26,474 - octoprint.plugins.rgb_status - INFO - Starting Idle Effect
2020-06-29 23:02:26,486 - octoprint.plugins.rgb_status - INFO - Putting KILL code in queue
2020-06-29 23:02:26,503 - octoprint.plugins.rgb_status - INFO - Joining Effect process
2020-06-29 23:02:26,520 - octoprint.plugins.rgb_status - INFO - Killed effect: Progress
2020-06-29 23:02:26,527 - octoprint.plugins.rgb_status - INFO - Starting new effect Solid Color
2020-06-29 23:02:26,538 - octoprint.plugins.rgb_status - INFO - Started new effect <Process(Solid Color, started)>
2020-06-29 23:02:26,600 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-06-29 23:02:26,603 - octoprint.plugins.rgb_status - INFO - Starting Fail Effect
2020-06-29 23:02:26,607 - octoprint.plugins.rgb_status - INFO - Putting KILL code in queue
2020-06-29 23:02:26,611 - octoprint.plugins.rgb_status - INFO - Joining Effect process
2020-06-29 23:02:26,635 - octoprint.plugins.rgb_status - INFO - Killed effect: Solid Color
2020-06-29 23:02:26,641 - octoprint.plugins.rgb_status - INFO - Starting new effect Pulse
2020-06-29 23:02:26,654 - octoprint.plugins.rgb_status - INFO - Started new effect <Process(Pulse, started)>
2020-06-29 23:02:33,868 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:XXXXXXXXXXX
2020-06-29 23:06:44,660 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2020-06-29 23:06:53,439 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2020-06-29 23:06:53,520 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2020-06-29 23:11:42,635 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2020-06-29 23:11:48,647 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2020-06-29 23:11:54,659 - octoprint.util.comm - INFO - No response from printer after 3 consecutive communication timeouts, considering it dead.
2020-06-29 23:11:54,674 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2020-06-29 23:11:54,686 - octoprint.plugins.rgb_status - INFO - Starting Disconnected Effect
2020-06-29 23:11:54,694 - octoprint.plugins.rgb_status - INFO - Putting KILL code in queue
2020-06-29 23:11:54,699 - octoprint.plugins.rgb_status - INFO - Joining Effect process
2020-06-29 23:11:54,733 - octoprint.plugins.rgb_status - INFO - Killed effect: Pulse
2020-06-29 23:11:54,740 - octoprint.plugins.rgb_status - INFO - Starting new effect Solid Color
2020-06-29 23:11:54,755 - octoprint.plugins.rgb_status - INFO - Started new effect <Process(Solid Color, started)>
2020-06-29 23:14:25,540 - octoprint.server.heartbeat - INFO - Server heartbeat <3
`

As you can see, it looks like it is working, but physically, no LEDs are lighting up.

cp2004, 2020-07-05:
head on over to your /boot/config.txt, and set
```
core_freq_min=500
core_freq=500
```
The Pi4 has variable clock speed, so this is needed to stop it from clocking-down when not in use. Not accounted for in the plugin yet.

clevere12, 2020-07-05:
Thanks, Yes, did that. 
I also ended up putting a different power supply on it the Pi.
The one that comes with it, I could not get anything to go.
The official Pi power supply, it works, as long as the USB-C connector is in the right orientation. The LEDs will come on
and function properly, but they flicker, even with the core_freq_min=500 and core_freq=500 set.

I have a much larger power supply connected to it now and things are working without a hitch. , one that will delivery 5v and up to 10amp. I believe the addition of the Fan running off the 3.3v pins may be drawing to much power to supply the 5v pins and the 3v pins at the same time.

cp2004, 2020-07-05:
Generally I wouldn't recommend powering much more than a handful of LEDs from the Pi only, especially if you have a webcam as well.

clevere12, 2020-07-05:
Agreed.
It has one very small fan, I don't know the specs and an Adafruit 1426  NeoPixel Stick - 8 x 5050 RGB LED with Integrated Drivers. 

The Pi3 doesn't have the fan, doesn't need it. I believe that is where the problem lies.

closed 2020-07-14 by EricHigdon