EricHigdon/OctoPrint-RGB_status #81: Forced to fill in details I don't know - Plugin install has broken my Octoprint!

MakeTheFuture, 2020-07-14:
Who on earth thought it was a good idea to force a user to input details they don't even know/have yet on reload after install of plugin? How do i disable/remove this terribly designed plugin now?

[image]

EricHigdon, 2020-07-14:
I’m not sure what details you are referring to. You should know your own password. If you don’t, it’s probably the default “raspberry” as the setup says (twice). In which case, all you have to do is click the blue buttons. 

It’s quite unlikely that you’ve accurately assessed the plugin as “terribly designed” when you haven’t progressed past the first step in setting it up. 

Do you have a constructive suggestion to input? I’d like to hear where you got hung up in the process. 

If you really want to just uninstall the plugin without giving it a try (totally fine), it’s the same process as any other OctoPrint plugin: https://community.octoprint.org/t/i-installed-a-broken-plugin-how-do-i-uninstall-it-again/217

mon5termatt, 2020-07-24:
If you dont know your own password maybe you shouldnt be using octoprint, or any linux based device for that matter...

Probably could close this "issue" btw

EricHigdon, 2020-07-24:
@mon5termatt thanks. I wanted to give some time to give actual input or make a request. I think it’s fair to say there has been plenty of time.

closed 2020-07-24 by EricHigdon

MakeTheFuture, 2020-07-24:
Of it's quite terribly designed. You can argue a strawman all you want and say stupid shit like 'if you don't know your password'. 

The point is, even if you enter the password your Octoprint install is fucked because of this plugin forcing you to enter these fields before proceeding. Your stuck in a chicken and egg situation that you cannot proceed past. Close the issue if you want but i guaranteeing there will be a lack of use because of what I'm highlighting here. If you can't see that then you just confirm my point that the design is broken.

No fields should require entry after restart, they should all be required before the plugin runs. A simple design change and quite obvious had it been designed properly to begin with.

MakeTheFuture, 2020-07-24:
Grats on being one of three plugins in the plugin repo that if a user installs (by accident or before they have all the 'required' fields to enter), it will brick their Octopi with only the manual fix via command line possible. For most users that means, un-fixable.

EricHigdon, 2020-07-24:
I’m not sure what part of my previous comments makes you think I “can’t see past my own chin”? I’m trying to get productive information about the issue you are having. The fields that are required are for your password. Hence the comment about knowing your password and suggestion for if you don’t know it.

I’m using OctoPrint’s built-in UI for the setup wizard. It pops up after a restart because the plugin runs continuously in a background process. Therefore, these fields are required before the plugin runs. 

There definitely is room for improvement in many areas of this plugin. I just don’t always have time to find and fix everything. If you have suggestions, I’m all ears. If you have a pull request, even better.

EricHigdon, 2020-07-24:
“ This should require the user to enter into the settings area and configure the fields and only place a notification on login to state the plugin is not executing because setup is still required.”

Not a bad idea, but I’d want it to be VERY clear why the plugin isn’t working properly. I created #83 to at least look into options here. Thanks for your input.

MakeTheFuture, 2020-07-24:
Righto, I had you pegged worse then you appeared to me at first. Did not expect you to open another issue. Thanks from the community for that. Thought you were just, yet another, maintainer incapable of handling any critical feedback unless it was given to them as they are the master of the universe and infallible (as you might be able to tell, it's the only type ive run into). I should have started a little more gentle/neutral, sorry about that.