EricHigdon/OctoPrint-RGB_status #82: (IDEA) Ability to Change "progress bar" animation

mon5termatt, 2020-07-24:
I would like to have it so that when im printing it displays lights all on white except one

0=lit -=dim
`10% 0-00000000000000000
20% 000-000000000000000
30% 00000-0000000000000
40% 0000000-00000000000`
ETC....

mon5termatt, 2020-07-24:
i would love to code this myself but i have no idea where to even begin

EricHigdon, 2020-07-24:
@mon5termatt this is a cool idea for an alternate Progress look. I think I understand, but I’m going to recreate your diagram here using 10 leds as a note to myself. 

`
10%:  -000000000
50%:  0000-00000
100%: 000000000-
`

mon5termatt, 2020-07-29:
yeah sorry for the crap formatting, code blocks are a bitch

mon5termatt, 2020-07-29:
also adjustable width would be nice for the people like me who are running 144 led's/meter strands

EricHigdon, 2020-07-29:
Right. It would be all percentage base like the current progress effect.

Note to self: this could be accomplished with a “highlight” checkbox that dims the current percentage pixel, while still using the base color and progress color settings. That would allow for a lot of flexibility.