EricHigdon/OctoPrint-RGB_status #89: Plugin stuck in setup mode after upgrading OctoPrint to python 3 / group gpio fpr user pi not recognized

konkz, 2020-11-01:
Hi there,
I upgraded OctoPrint to python 3 following https://github.com/cp2004/OctoPrint-Upgrade-To-Py3
Everything worked fine,  but the RGB-Status plugin is now stuck in setup mode:

[image]

The user pi is member of the group gpio:

[image]

Just following the setup wizard by clicking "Add gpio Group" doesn't work either. The problem is reproducible. Tried several times installing, uninstalling, rebooting ...

Plugin version is 1.7.2, octoprint version is 1.4.2. I don't use OctoPi but installed Octoprint myself on Raspberry OS lite 2020-08-20.

Btw. uninstalling RGB-Status plugin helped me, being able to use Octoprint again: https://community.octoprint.org/t/i-installed-a-broken-plugin-how-do-i-uninstall-it-again/217

`/home/pi/OctoPrint/venv/bin/pip uninstall RGB-Status`
is the command if you installed octoprint yourself following https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

I hope you can fix this issue.

cheers

cp2004, 2020-11-01:
This is likely a Python 2/3 issue, since the default encoding of string literals has changed - the plugin would need to decode the output. There are other options for LEDs on your printer, https://plugins.octoprint.org/search/?q=ws281x

@EricHigdon when you get time to fix this problem, you likely have to add `.decode('utf-8')` to the process output, and add `from __future__ import unicode_literals`

MaikelKelder, 2020-11-29:
I have exact the same issue, no fix yet?? 
I am using the 0.18 since I need to for my pi4

cp2004, 2020-11-29:
@MaikelKelder indeed, there has not yet been a fix. You can't use this plugin on Python 3 properly, and must look at the other options. 
@EricHigdon This is going to be more of a problem as more and more people upgrade, might be worth taking a look (I provided the likely solution above... 🙂 )

MaikelKelder, 2020-11-29:
@cp2004 I guess I have a bigegr problem, I cant even get my LEDs to work on my new RPI4 with 0.18 Octopi.
Tried everything I would think, on my RPI3B no issues, I think the GPIO pins are not really working in 0.18....
So need to wait for a fix anyhow.

cp2004, 2020-11-29:
I also have that issue reported on my WS281x plugin, but yet to test it out. I fear it may be an upstream library issue.

MaikelKelder, 2020-11-29:
Yes, I tried all of them, nothing works, its a Octopi thing I am suspecting. Tried different GPIO pins, etc... separate power, etc.
So it doenst seem like the plugin is at fault here...

Lisheng2016, 2020-12-07:
> @cp2004 I guess I have a bigegr problem, I cant even get my LEDs to work on my new RPI4 with 0.18 Octopi.
> Tried everything I would think, on my RPI3B no issues, I think the GPIO pins are not really working in 0.18....
> So need to wait for a fix anyhow.

How do you wire the led?I forgot to use common ground in the past and that was the cause.

MaikelKelder, 2020-12-07:
Sure I did, tried all the options in the book, even changing status of the
gpio pin, alternate mode of the pin, etc.
Just leds don’t react, I will try a led ring later but for now I stopped
trying.
Even tried a second pi4, same issue.

Cant seem to pinpoint the problem.