EricHigdon/OctoPrint-RGB_status #95: Installer stuck on adding PI to GPIO group.

dbreggin, 2021-02-26:
From other work I had already done, pi was already in the GPIO group.  It fails trying to add a user (pi) which is already there.  There is no graceful way out of this endless loop of programatic tail chasing.

Please catch this condition and allow the installer to continue.

JimKnightster, 2021-02-28:
Same issue here.  Question is how do I get past this so I can load my Octoprint?  I cant get past this 'required' page.

cp2004, 2021-02-28:
Head to `http://<the-ip>/recovery`, start in safe mode and then remove the plugin.

mikekscholz, 2021-04-29:
For anyone else that may be experiencing this, update PSU Control if youre using it because it was causing all install wizards to lock up.

cp2004, 2021-04-29:
Installing this plugin from the plugin manager in it's current state still breaks things as far as I know, we were recieving a lot of support issues around the OctoPrint forums & discord. PSU control did not break wizards, it was Spool Manager in that case.

The issue here is that your fixes have not been released - the URL submitted to the plugin manager is for the master branch, and your fix was merged to main.

jes1510, 2022-09-05:
Same issue here.  Finally uninstalled it in safe mode and installed the WS281x LED Status plugin which worked flawlessly.  This project looks like it may be abandoned.

danielgardner2004, 2023-01-26:
how do i get into safe mode? as soon as octoprint loads it goes to the setup screen in milliseconds